Skaginn3x/framework #12: confman further development

jbbjarnason, 2023-03-29:
- Log configuration changes
- Make configuration management available on json-rpc ( cockpit f.e. )
- Services declare configuration fields, type, **limits** and defaults to configuration manager over ipc.

omarhogni, 2023-10-11:
Points 2 & 3 addressed already. Issue updated to reflect that.

jbbjarnason, 2023-10-12:
We could fix the disk writing problem in parallel. And maybe skip the part of logging, my suggestion would be to keep 10 copies of old config.

1. Write to file function would create file `<config_name>.json.tmp`
2. `rm` file 10
3. mv files 1-9 to 2-10 (`<config_name>.json.1) etc
4. cp file  `<config_name>.json` to  `<config_name>.json.1`
5. mv file `<config_name>.json.tmp` to `<config_name>.json`

The tmp file could be used as lock file, if we would like to detect failure in this process.

Lastly 10 could be default but this could be an argument or change-able parameter in confman API.

For frontend, populate a list of timestamp of those 10 files. Frontend can call a method with argument integer with indices of the timestamp to fetch and compare in UI.

omarhogni, 2023-10-20:
I think this is allot of work.
Can we wait with removing config and when writing a new config just write the old as well timestamped?

omarhogni, 2023-10-20:
Idea proposed in #218

omarhogni, 2023-10-20:
@magni-mar Can you implement the configuration retention?

magni-mar, 2023-10-20:
I can start working on this issue, can not promise any deadline though as I have no idea how long this will take.