Skaginn3x/framework #117: json schema research

jbbjarnason, 2023-07-05:
referring to [json schema diealect guideline](https://json-schema.org/understanding-json-schema/reference/schema.html#guidelines), custom keywords can be made to declare more meta data outside the standard.

here are some useful information in terms of units, https://github.com/frictionlessdata/specs/issues/216

the proposal is the following

make a struct containing optional standard schema attribute keywords
```c++
struct  schema_attributes {
std::optional<std::string_view> description{};
std::optional<std::int64_t> min{};
std::optional<std::int64_t> max{};
std::variant<std::monostate, std::string_view, std::int64_t, std::uint64_t, double, bool> default{};
// ...
};
```

make a container similar to `glz::object`, like `tfc::json::object` 

```c++
struct test {
int a{};
struct glaze {
static constexpr value{  tfc::json::object("a", &test::a, schema_attributes{ .description="length to sensor",.min=0,.max=300,.default=20 } ) };
};
};
```

omarhogni, 2023-10-11:
Can this be closed as complete @jbbjarnason ?

jbbjarnason, 2023-10-12:
cloesed in favor of https://github.com/Skaginn3x/framework/pull/133

closed 2023-10-12 by jbbjarnason