Skaginn3x/framework #145: Add title to variant types in json schema

jbbjarnason, 2023-08-29:
See `oneOf` it needs to indicate title of each option

```json
{
    "type": [
        "object"
    ],
    "properties": {
        "config": {
            "$ref": "#/$defs/with_variant"
        }
    },
    "additionalProperties": false,
    "$defs": {
        "enum_thingy": {
            "type": [
                "string"
            ],
            "oneOf": [
                {
                    "const": "foo"
                },
                {
                    "const": "bar"
                }
            ]
        },
        "int32_t": {
            "type": [
                "integer"
            ],
            "minimum": -2147483648,
            "maximum": 2147483647
        },
        "std::string": {
            "type": [
                "string"
            ]
        },
        "std::variant<std::monostate,option_1,option_2>": {
            "type": [
                "number",
                "string",
                "boolean",
                "object",
                "array",
                "null"
            ],
            "oneOf": [
                {
                    "type": [
                        "null"
                    ]
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "amper": {
                            "$ref": "#/$defs/units::quantity<electric_current,dA,uint16_t>",
                            "description": "amper description"
                        }
                    },
                    "additionalProperties": false
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "a": {
                            "$ref": "#/$defs/std::string",
                            "description": "A description"
                        },
                        "sec": {
                            "$ref": "#/$defs/tfc::observable<std::chrono::duration<glz::unknown,std::nano>>",
                            "description": "sec description"
                        }
                    },
                    "additionalProperties": false
                }
            ]
        },
        "tfc::observable<std::chrono::duration<glz::unknown,std::nano>>": {
            "type": [
                "integer"
            ],
            "minimum": -9223372036854775808,
            "maximum": 9223372036854775807,
            "x-tfc": {
                "unit": "ns",
                "dimension": "time",
                "ratio": {
                    "numerator": 1,
                    "denominator": 1000000000
                },
                "required": true
            }
        },
        "units::quantity<electric_current,dA,uint16_t>": {
            "type": [
                "integer"
            ],
            "minimum": 0,
            "maximum": 65535,
            "x-tfc": {
                "unit": "dA",
                "dimension": "electric_current",
                "ratio": {
                    "numerator": 1,
                    "denominator": 10
                },
                "required": true
            }
        },
        "with_variant": {
            "type": [
                "object"
            ],
            "properties": {
                "a_int": {
                    "$ref": "#/$defs/int32_t",
                    "description": "A int description"
                },
                "thingy": {
                    "$ref": "#/$defs/enum_thingy"
                },
                "variant": {
                    "$ref": "#/$defs/std::variant<std::monostate,option_1,option_2>",
                    "description": "variant description"
                }
            },
            "additionalProperties": false
        }
    }
}
```

jbbjarnason, 2023-10-12:
@SindriTh I think I fixed this, but I don't remember where or if. Was this fixed?

SindriTh, 2023-10-12:
> @SindriTh I think I fixed this, but I don't remember where or if. Was this fixed?

The DBUS filter schema shows:

"oneOf": [
        {
          "type": [
            "number",
            "string",
            "boolean",
            "object",
            "array",
            "null"
          ],
          "title": "tfc::ipc::filter::invert"
        },
        {
          "type": [
            "object"
          ],
          "properties": {
            "time_off": {
              "$ref": "#/$defs/std::chrono::duration<int64_t,std::milli>",
              "description": "Falling edge settling delay, applied on next event, NOT current one if already processing"
            },
            "time_on": {
              "$ref": "#/$defs/std::chrono::duration<int64_t,std::milli>",
              "description": "Rising edge settling delay, applied on next event, NOT current one if already processing"
            }
          },
          "additionalProperties": false,
          "title": "tfc::ipc::filter::timer"
        }
      ]
      
      So I think this has been implemented. I'm not sure which merge though

SindriTh, 2023-10-12:
I believe it was #150 @jbbjarnason

jbbjarnason, 2023-10-12:
closed in favor of https://github.com/Skaginn3x/framework/pull/150

closed 2023-10-12 by jbbjarnason