Skaginn3x/framework #243: import tfc::testing

magni-mar, 2023-11-03:
when linking to library tfc::testing cmake throws an error

note I am using tfc-framework 2023.11.1

```bash
CMake Error at projects/falfish/weighing/tests/CMakeLists.txt:11 (target_link_libraries):
  Target "weighing_system_integration_tests" links to:

    tfc::testing

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.
```

jbbjarnason, 2024-01-25:
this if clause should be removed, https://github.com/Skaginn3x/framework/blob/main/libs/CMakeLists.txt#L1

magni-mar, 2024-03-22:
this has been resolved

closed 2024-03-22 by magni-mar