Skaginn3x/framework #351: array of item description not shown

omarhogni, 2023-12-28:
When working with arrays of items the description designated for that array is nowhere to be found.

SindriTh, 2024-01-21:
Could you provide a sample array, along with screenshots?

omarhogni, 2024-01-31:
When arrays are configured with a description the description is not visible anywhere.
[image]

Expected description string
[image]

The description text appears to follow the type decleration
```json
      "infeed_chain_to_sorter": {
        "$ref": "#/$defs/std::vector<tfc::observable<std::variant<std::monostate,printing_motor,atv320>>>",
        "description": "Infeed conveyors before the sorter"
      }
```

See for config.
[config_example.json](https://github.com/Skaginn3x/framework/files/14114578/config_example.json)

omarhogni, 2024-02-01:
Another issue with this same config the website cannot render std::size_t we can create another issue if you want but the config structure is the same as here.
[image]

SindriTh, 2024-02-14:
Do we want Array descriptions to show below the title:
[image]

or on hover like shown in "sec"?:
[image]

@omarhogni

PR #477 (2024-02-14): General Configurator fixes
fixes #351 - Implementation might change
fixes #462 - Fixed "0" handling
changed implementation of #354 - Requires testing

This PR is a mixed bag of minor fixes
Current implementation of #351 looks slightly off, open to suggestions on alternative implementations
closed 2024-02-14 by jbbjarnason