Skaginn3x/framework #354: [cockpit] ascending/descending order of items

magni-mar, 2024-01-09:
Would be nice to have elements in cockpit sorted by ascending/descending numbers, letters and such in the Configurator.

[image]

SindriTh, 2024-01-23:
This is currently being sorted alphabetically, I can split the string in such a way that it has "atv320_i" which is sorted alphabetically, then "6", "7" etc which is sorted as a number. Would this cover all cases?

jbbjarnason, 2024-01-23:
@SindriTh  I am a bit confused, do numbers not get sorted while being a part of the name,

Here is another example:

[image]

SindriTh, 2024-01-23:
I will look into this.

However, please note that
```
atv320_i7
atv320_i8
atv320_i9
atv320_i10
```
would alphabetically sort to 
```
atv320_i10
atv320_i7
atv320_i8
atv320_i9
```
due to 1 being "earlier in the alphabet" than 7.

This is clearly not the issue here but will be something that we will have to solve

jbbjarnason, 2024-02-12:
this is better but still weird

[image]

PR #392 (2024-02-08): Sorted config items in sidebar
Items in sidebar are now sorted.

[image]

closes #354
closed 2024-02-16 by SindriTh