Skaginn3x/framework #361: [cockpit] checkbox instead of dropdown.

magni-mar, 2024-01-11:
Would be nice to have a easier way of choosing which signals to publish in mqtt bridge. In real life an engineer would likely want to see most of the signals, so checkboxes or something similar would be better than a dropdown as it takes a really long time to choose all of the signals.

[image]

jbbjarnason, 2024-01-23:
I think it is hard to deduce by the frontend how to display the data differently in special cases, I would suggest being able to specify by json schema or UI schema declaring which widget would be appropriate to use. Then the responsibility is always part of the backend.

SindriTh, 2024-01-23:
I agree with @jbbjarnason ,this issue should be addressed by the backend,