Skaginn3x/framework #385: Issue with Rendering JSONSchema Due to Special Characters in Reference Names

SindriTh, 2024-01-21:
### Description

We are encountering a rendering issue in our application related to processing a JSONSchema. The specific problem arises with the reference name that contains special characters (specifically a slash `/`). This issue is evident in the reference to `tfc::observable<units::quantity<speed,nm/s,int64_t>>`, where the slash in `nm/s` seems to interfere with the proper indexing and parsing of the schema.

#### Details

- **Schema Portion with Issue**:
  ```json
  "tfc::observable<units::quantity<speed,nm/s,int64_t>>": {
      "type": ["integer"],
      "minimum": -9223372036854775808,
      "maximum": 9223372036854775807,
      "x-tfc": {
          "unit": {
              "unit_ascii": "nm/s",
              "unit_unicode": "nm/s"
          },
          "dimension": "speed",
          "required": true
      }
  }
  ```

- **Problematic Reference**:
  ```json
  "$ref": "#/$defs/tfc::observable<units::quantity<speed,nm/s,int64_t>>",
  ```

#### Impact

The presence of the slash (`/`) in the reference name seems to be causing issues with the JSONSchema rendering in our application. This is likely due to the way JSON pointers are interpreted, where a slash typically signifies a new nested level in the JSON object.

#### Possible Solutions

1. **Encoding Special Characters**: Consider encoding special characters in the reference names. For example, replacing `/` with `%2F` might resolve the issue.

2. **Schema Adjustment**: If possible, adjust the schema to avoid using special characters in reference names.

3. **Custom Parsing Logic**: Implement custom parsing logic in the application to correctly handle these special characters in reference names.

SindriTh, 2024-01-22:
Please advise @jbbjarnason

jbbjarnason, 2024-01-23:
@SindriTh we should not support something that does not make sense, let's remove slashes from defs on the backend side. 
Furthermore, statically assert in backend, so it won't compile if this occurs again.

closed 2024-01-29 by omarhogni