Skaginn3x/framework #390: Investigate ethercat coredump

jbbjarnason, 2024-01-23:
Core is available to download from grader in falfish, and occured for this hash 49a32716c3e5fc71a32098515412a66e941db1b2

omarhogni, 2024-01-24:
This is difficult to reproduce. Needs further investigation if we manage to reproduce.

jbbjarnason, 2024-01-24:
Attained information during the past days:

Dump:
Thu 2024-01-18 10:32:28 GMT  5320  999  991 SIGSEGV missing  /usr/local/bin/ethercat           

Inspect in clion:
[image]

Crash occurs here, check the boggled pointer address of handler_:
[image]

[ethercat.log](https://github.com/Skaginn3x/framework/files/14037829/ethercat.log)