Skaginn3x/framework #455: Variant property with name

jbbjarnason, 2024-02-12:
Currently variant properties show up as choose variant instead of the actual property name, example:

[image]

```json
(<('{
    "discharge delay": 1000,
    "minimum discharge duration": 1000,
    "await sensor timeout": 60000,
    "run on discharge": true,
    "allow item removal": false,
    "motor": null
}', '{
    "type": [
        "object"
    ],
    "properties": {
        "config": {
            "$ref": "#/$defs/tfc::sensor_control"
        }
    },
    "additionalProperties": false,
    "$defs": {
        "bool": {
            "type": [
                "boolean"
            ]
        },
        "std::chrono::duration<int64_t,std::milli>": {
            "type": [
                "integer"
            ],
            "minimum": -9223372036854775808,
            "maximum": 9223372036854775807,
            "x-tfc": {
                "unit": {
                    "unit_ascii": "ms",
                    "unit_unicode": "ms"
                },
                "dimension": "time",
                "ratio": {
                    "numerator": 1,
                    "denominator": 1000
                },
                "required": true
            }
        },
        "std::optional<std::chrono::duration<int64_t,std::milli>>": {
            "type": [
                "integer"
            ],
            "minimum": -9223372036854775808,
            "maximum": 9223372036854775807,
            "x-tfc": {
                "unit": {
                    "unit_ascii": "ms",
                    "unit_unicode": "ms"
                },
                "dimension": "time",
                "ratio": {
                    "numerator": 1,
                    "denominator": 1000
                },
                "required": false
            }
        },
        "std::optional<units::quantity<speed,double>>": {
            "type": [
                "number"
            ],
            "minimum": -1.7976931348623157E308,
            "maximum": 1.7976931348623157E308,
            "x-tfc": {
                "unit": {
                    "unit_ascii": "mm/s",
                    "unit_unicode": "mm/s"
                },
                "dimension": "speed",
                "required": false
            }
        },
        "std::shared_ptr<tfc::observable<std::variant<std::monostate,printing_motor,atv320,stub DONT PICK>>>": {
            "type": [
                "number",
                "string",
                "boolean",
                "object",
                "array",
                "null",
                "null"
            ],
            "oneOf": [
                {
                    "type": [
                        "null"
                    ],
                    "title": "std::monostate",
                    "const": null
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "name": {
                            "$ref": "#/$defs/tfc::observable<std::string>",
                            "description": "Name of the motor"
                        },
                        "nominal": {
                            "$ref": "#/$defs/std::optional<units::quantity<speed,double>>",
                            "description": "Speed of virtual motor in physical quantities while at 50Hz"
                        }
                    },
                    "additionalProperties": false,
                    "title": "printing_motor"
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "slave_id": {
                            "$ref": "#/$defs/tfc::observable<uint16_t>"
                        }
                    },
                    "additionalProperties": false,
                    "title": "atv320"
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "name": {
                            "$ref": "#/$defs/std::string_view"
                        }
                    },
                    "additionalProperties": false,
                    "title": "stub DONT PICK"
                }
            ]
        },
        "std::string": {
            "type": [
                "string"
            ]
        },
        "std::string_view": {
            "type": [
                "string"
            ]
        },
        "tfc::observable<std::string>": {
            "type": [
                "string"
            ]
        },
        "tfc::observable<uint16_t>": {
            "type": [
                "integer"
            ],
            "minimum": 0,
            "maximum": 65535
        },
        "tfc::sensor_control": {
            "type": [
                "object"
            ],
            "properties": {
                "allow item removal": {
                    "$ref": "#/$defs/bool",
                    "description": "Allow removing item while awaiting discharge, this will remove the item from the stream.That means iff item is removed from sensor the conveyor will run to try find the missing item until sensor timeout"
                },
                "await sensor timeout": {
                    "$ref": "#/$defs/std::chrono::duration<int64_t,std::milli>",
                    "description": "Timeout for awaiting sensor input."
                },
                "discharge delay": {
                    "$ref": "#/$defs/std::optional<std::chrono::duration<int64_t,std::milli>>",
                    "description": "Delay after falling edge of sensor to keep output of discharge active high."
                },
                "minimum discharge duration": {
                    "$ref": "#/$defs/std::chrono::duration<int64_t,std::milli>",
                    "description": "Minimum discharge duration, set with relation to item length and conveyor speed."
                },
                "motor": {
                    "$ref": "#/$defs/std::shared_ptr<tfc::observable<std::variant<std::monostate,printing_motor,atv320,stub DONT PICK>>>",
                    "description": "The motor to control"
                },
                "run on discharge": {
                    "$ref": "#/$defs/bool",
                    "description": "Run the motor when discharging an item. Useful for corner conveyors, or where lift is used to discharge, not the motor itself."
                }
            },
            "additionalProperties": false
        }
    }
}')>,)
```

SindriTh, 2024-02-14:
The current implementation is as follows:
```js
 <InputLabel>{schema.get('title') as string ?? 'Choose Variant'}</InputLabel>
```

for this to work the following changes need to be made:

```json
        "std::shared_ptr<tfc::observable<std::variant<std::monostate,printing_motor,atv320,stub DONT PICK>>>": {
            "type": [
                "number",
                "string",
                "boolean",
                "object",
                "array",
                "null",
                "null"
            ],
            **title: "Variant Name",**
            "oneOf": [
                {
                    "type": [
                        "null"
                    ],
                    "title": "std::monostate",
                    "const": null
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "name": {
                            "$ref": "#/$defs/tfc::observable<std::string>",
                            "description": "Name of the motor"
                        },
                        "nominal": {
                            "$ref": "#/$defs/std::optional<units::quantity<speed,double>>",
                            "description": "Speed of virtual motor in physical quantities while at 50Hz"
                        }
                    },
                    "additionalProperties": false,
                    "title": "printing_motor"
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "slave_id": {
                            "$ref": "#/$defs/tfc::observable<uint16_t>"
                        }
                    },
                    "additionalProperties": false,
                    "title": "atv320"
                },
                {
                    "type": [
                        "object"
                    ],
                    "properties": {
                        "name": {
                            "$ref": "#/$defs/std::string_view"
                        }
                    },
                    "additionalProperties": false,
                    "title": "stub DONT PICK"
                }
            ]
        },
```

I fail to see where else I would find the property name.

jbbjarnason, 2024-02-17:
> The current implementation is as follows:
>
> ```js
>  <InputLabel>{schema.get('title') as string ?? 'Choose Variant'}</InputLabel>
> ```
>
> for this to work the following changes need to be made: [...]

@SindriTh the variant has a property name, `motor` in this case which should be checked. For example if there is two properties with the same reference type but two distinct property names.

SindriTh, 2024-02-17:
I will look into grabbing the parent element’s property name and using that

PR #510 (2024-02-21): Changed label on variant
Changed variant widget in configurator to check for variant name in the following order:

- Parent schema's property name
- "title" property

Defaults to "Choose variant"

Closes #455
closed 2024-02-21 by jbbjarnason