Skaginn3x/framework #511: Investigate SDO malfunction

jbbjarnason, 2024-02-21:


jbbjarnason, 2024-02-21:
Tried reproducing by sending changes to sdo on all parameters for one motor, did not occur.

The problem was most likely that the ethercat exe got to many ping pong requests on motors and was unable to keep the cycle running stably. However, @omarhogni found that the reset might need to be pulsed, this wasn't verified nor implemented. @omarhogni can you put a link or screenshot for reference? and close as we cannot reproduce.

omarhogni, 2024-03-25:
the [ATV320 Programming Manual](https://github.com/Skaginn3x/framework/files/14741855/ATV320_Programming_Manual_EN_NVE41295_04.pdf)
[image]

There is mention of a value change for the reset bit from low to high.

The case i was talking about was 
- Our reset bit is high
- a auto resetable fault occurs
- the fault does not reset because the reset bit was already high and now we hold it high.