Skaginn3x/framework #538: Look into reading analog inputs from ATV320 periodically or by increasing the PDO size

omarhogni, 2024-04-15:


omarhogni, 2024-04-15:
This is not possible

[image]

omarhogni, 2024-04-15:
@jbbjarnason How do you want to proceed?

jbbjarnason, 2024-04-15:
The options I see valid:
- Accept sdo read being blocking and poll on interval greater than 1s, and hope for the best 
- Somehow implement async sdo read in soem
- Use Rust
- Use Lenze

I think the least work would be using Lenze.
I would like to research how much effort adding async functionality to soem before answering.

omarhogni, 2024-04-15:
Let's plan on using lenze for now, I will close this issue as complete and add a task to implement a proper lenze driver inside of tfc.

closed 2024-04-15 by omarhogni